khoih-prog/TimerInterrupt #1: ISR_RPM_Measure Example crash

csaba911, 2020-06-12:
```
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Core 1 register dump:
PC      : 0x400813eb  PS      : 0x00050234  A0      : 0x4008489c  A1      : 0x3ffbe7b0  
A2      : 0x00000000  A3      : 0x00000002  A4      : 0x000000ff  A5      : 0x4008a720  
A6      : 0x3ffbebfc  A7      : 0x00000000  A8      : 0x3ff60000  A9      : 0x3ff5f000  
A10     : 0xb4953222  A11     : 0xb4953222  A12     : 0x3ffc04c8  A13     : 0x00001387  
A14     : 0xe00a1000  A15     : 0x0000060e  SAR     : 0x00000010  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xffffffff  
Core 1 was running in ISR context:
EPC1    : 0x40081440  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x400813eb

Backtrace: 0x400813eb:0x3ffbe7b0 0x40084899:0x3ffbe7d0 0x4008143d:0x3ffb1f00 0x400d0e05:0x3ffb1f60 0x400d1bcf:0x3ffb1fb0 0x400884c1:0x3ffb1fd0

Core 0 register dump:
PC      : 0x400e99ae  PS      : 0x00060134  A0      : 0x800d4782  A1      : 0x3ffbbff0  
A2      : 0x00000000  A3      : 0x00000001  A4      : 0x00000000  A5      : 0x00000001  
A6      : 0x00060120  A7      : 0x00000000  A8      : 0x800d434a  A9      : 0x3ffbbfc0  
A10     : 0x00000000  A11     : 0x40085024  A12     : 0x00060120  A13     : 0x3ffbb970  
A14     : 0x00000000  A15     : 0x3ffbbce0  SAR     : 0x00000000  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000  

Backtrace: 0x400e99ae:0x3ffbbff0 0x400d477f:0x3ffbc010 0x400899b2:0x3ffbc030 0x400884c1:0x3ffbc050

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 188777542, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

Starting
ESP32TimerInterrupt: _timerNo = 1, _fre = 1000000.00, _count = 0 - 1
RPM = 0.00, rotationTime = 5000

```

khoih-prog, 2020-06-15:
Hi @csaba911 

Thanks for using the library, but you're either using the wrong library or post the issue in the wrong library.

This [**TimerInterrupt Library**](https://github.com/khoih-prog/TimerInterrupt) is designed only for certain supported **Arduino** boards in this list of [**Supported Arduino Boards**](https://github.com/khoih-prog/TimerInterrupt#supported-arduino-boards).

For **ESP32**, you have to use this [ISR_RPM_Measure example](https://github.com/khoih-prog/ESP32TimerInterrupt/blob/master/examples/ISR_RPM_Measure/ISR_RPM_Measure.ino) after  installing this [**ESP32TimerInterrupt Library**](https://github.com/khoih-prog/ESP32TimerInterrupt)

closed 2020-06-15 by khoih-prog

khoih-prog, 2020-06-15:
Hi @csaba911 

If you're usong the correct [**ESP32TimerInterrupt Library**](https://github.com/khoih-prog/ESP32TimerInterrupt), then please fix the typo in the  [ISR_RPM_Measure example](https://github.com/khoih-prog/ESP32TimerInterrupt/blob/master/examples/ISR_RPM_Measure/ISR_RPM_Measure.ino) as follows:

From

```
// Interval in microsecs
  if (ITimer1.attachInterruptInterval(TIMER1_INTERVAL_MS, TimerHandler1))
    Serial.println("Starting  ITimer1 OK, millis() = " + String(millis()));
  else
    Serial.println("Can't set ITimer1. Select another freq., duration or timer");
```

To

```
// Interval in microsecs, must multiply 1000
  if (ITimer1.attachInterruptInterval(TIMER1_INTERVAL_MS * 1000, TimerHandler1))
    Serial.println("Starting  ITimer1 OK, millis() = " + String(millis()));
  else
    Serial.println("Can't set ITimer1. Select another freq., duration or timer");
```

khoih-prog, 2020-06-15:
Hi @csaba911 

Just fix the typo in the  [ISR_RPM_Measure example](https://github.com/khoih-prog/ESP32TimerInterrupt/blob/master/examples/ISR_RPM_Measure/ISR_RPM_Measure.ino) of [**ESP32TimerInterrupt Library**](https://github.com/khoih-prog/ESP32TimerInterrupt).

Thanks,