khoih-prog/TimerInterrupt #3: Arduino Uno Wifi rev2 compatibility

tomasskratochvil, 2020-10-05:
Good morning everyone,

I have tried to compile this library for Arduino Uno Wifi rev2 with ATmega4809 uCPU. I would like to kindly ask you, if you can add support for Arduino Uno Wifi rev2. Thank you very much. The report from compilation is attached bellow:

Arduino: 1.8.13 (Windows 10), Vývojová deska: "Arduino Uno WiFi Rev2, ATMEGA328"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\tomas.kratochvil\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=on -ide-version=10813 -build-path C:\Users\TOMAS~1.KRA\AppData\Local\Temp\arduino_build_74491 -warnings=none -build-cache C:\Users\TOMAS~1.KRA\AppData\Local\Temp\arduino_cache_356696 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\tomas.kratochvil\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino

C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\tomas.kratochvil\Documents\Arduino\libraries -fqbn=arduino:megaavr:uno2018:mode=on -ide-version=10813 -build-path C:\Users\TOMAS~1.KRA\AppData\Local\Temp\arduino_build_74491 -warnings=none -build-cache C:\Users\TOMAS~1.KRA\AppData\Local\Temp\arduino_cache_356696 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino17 -prefs=runtime.tools.arduinoOTA.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\arduinoOTA\1.3.0 -prefs=runtime.tools.avr-gcc.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\tools\avr-gcc\7.3.0-atmel3.6.1-arduino5 -verbose C:\Users\tomas.kratochvil\Documents\Arduino\sketch_oct05a\sketch_oct05a.ino

Using board 'uno2018' from platform in folder: C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6

Using core 'arduino' from platform in folder: C:\Users\tomas.kratochvil\AppData\Local\Arduino15\packages\arduino\hardware\megaavr\1.8.6

Detecting libraries used...

"C:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\uno2018" "C:\\Users\\TOMAS~1.KRA\\AppData\\Local\\Temp\\arduino_build_74491\\sketch\\sketch_oct05a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

Alternatives for TimerInterrupt.h: [TimerInterrupt@1.0.2]

ResolveLibrary(TimerInterrupt.h)

  -> candidates: [TimerInterrupt@1.0.2]

"C:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\uno2018" "-IC:\\Users\\tomas.kratochvil\\Documents\\Arduino\\libraries\\TimerInterrupt\\src" "C:\\Users\\TOMAS~1.KRA\\AppData\\Local\\Temp\\arduino_build_74491\\sketch\\sketch_oct05a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

"C:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\uno2018" "-IC:\\Users\\tomas.kratochvil\\Documents\\Arduino\\libraries\\TimerInterrupt\\src" "C:\\Users\\tomas.kratochvil\\Documents\\Arduino\\libraries\\TimerInterrupt\\src\\ISR_Timer.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

"C:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino5/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega4809 -DF_CPU=16000000L -DARDUINO=10813 -DARDUINO_AVR_UNO_WIFI_REV2 -DARDUINO_ARCH_MEGAAVR -DUNO_WIFI_REV2_328MODE -DMILLIS_USE_TIMERB3 "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino/api/deprecated" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\cores\\arduino" "-IC:\\Users\\tomas.kratochvil\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\megaavr\\1.8.6\\variants\\uno2018" "-IC:\\Users\\tomas.kratochvil\\Documents\\Arduino\\libraries\\TimerInterrupt\\src" "C:\\Users\\tomas.kratochvil\\Documents\\Arduino\\libraries\\TimerInterrupt\\src\\TimerInterrupt.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

VAROVÁNÍ: knihovna TimerInterrupt je určena pro běh na architektuře avr a může být nekompatibilní s Vaší vývojovou deskou, která má architekturu megaavr.

Generating function prototypes...

[... cut ...]
C:\Users\tomas.kratochvil\Documents\Arduino\libraries\TimerInterrupt\src\TimerInterrupt.cpp: In member function 'void TimerInterrupt::set_OCR()':

C:\Users\tomas.kratochvil\Documents\Arduino\libraries\TimerInterrupt\src\TimerInterrupt.cpp:154:7: error: 'OCR1A' was not declared in this scope

       OCR1A = _OCRValueToUse;

       ^~~~~

Použití knihovny TimerInterrupt ve verzi 1.0.2 v adresáři: C:\Users\tomas.kratochvil\Documents\Arduino\libraries\TimerInterrupt 

exit status 1

Nastala chyba při kompilaci u desky Arduino Uno WiFi Rev2.

khoih-prog, 2020-10-05:
@tomasskratochvil 

Thanks for your interest in the library.
This library doesn't support Arduino **Uno Wifi rev2 (ATMega328 or ATMega4809)** and I'm sorry I don't have time to spend to add support to those boards and their popularity is very limited now.

You can do some research and find you if other libraries can support.

closed 2020-10-05 by khoih-prog

khoih-prog, 2021-04-02:
Hi @tomasskratochvil 

I'm just finished the initial (not sure it's working yet) version of **megaAVR_TimerInterrupt** for UNO WiFi Rev 2, but I don't have the UNO WiFi Rev 2 to test.

Are you willing to test that new library? If yes, I'll place it in private repository and invite you, so that you can download and test.

Regards,