khoih-prog/TimerInterrupt #5: Error compiling for Arduino Pro Micro

balucio, 2020-10-15:
I try your library on Arduino Pro Micro, i get following error:
```
Arduino/libraries/TimerInterrupt/src/TimerInterrupt.cpp:590:3: error: 'else' without a previous 'if'
   else if (_timer == 1)
   ^~~~
```

khoih-prog, 2020-10-15:
You have to check your installation. To use interrupt-related functions and features, **more hardware / firmware-related knowledge is necessary**.

The examples are written mostly for Mega using Serial3, Serial1, 5 timers. To apply to Nano/UNO/Pro Mini, you must know how to adjust.

**I just retest and compile OK with Arduino Pro Mini**

[image]

closed 2020-10-15 by khoih-prog

khoih-prog, 2020-10-15:
The board Arduino Micro, using ATmega32U4, is not supported by this library.

khoih-prog, 2021-04-03:
Arduino / Sparkfun /  Generic ATmega32u4, ATmega328P, etc. boards are now all supported.

---

### Releases v1.4.1

1. Add support to **Generic or Sparkfun AVR ATmega_32U4** such as **AVR_MAKEYMAKEY, AVR_PROMICRO, etc.**
2. Add support to **Generic or Sparkfun AVR ATmega_328(P)** such as **ARDUINO_REDBOT, ARDUINO_AVR_DIGITAL_SANDBOX, etc.**
3. Add support to **Generic or Sparkfun AVR ATmega128RFA1** such as **ATMEGA128RFA1_DEV_BOARD, etc.**

### Releases v1.4.0

1. Add support to **Adafruit AVR ATMEGA_32U4** such as **AVR_FLORA8, AVR_FEATHER32U4, etc.**
2. Add support to **Adafruit AVR ATMEGA_328(P)** such as **AVR_FEATHER328P, AVR_METRO, etc.**
3. Update examples

### Releases v1.3.0

1. Add support to **AVR ATMEGA_16U4, ATMEGA_32U4** such as **Leonardo, YUN, ESPLORA, etc.**
2. Update examples