khoih-prog/TimerInterrupt #7: Consider optimizing Flash and SRAM usage in the examples and readme sample code

Django0, 2020-12-17:
This is a wonderful library, good work!
If possible consider updating your examples with the use of the macro F(). https://learn.adafruit.com/memories-of-an-arduino/optimizing-sram
This might help others lowering their memory usage. Although this is just in the example code.
```
Serial.println("Starting  ITimer1 OK, millis() = " + String(millis()));`
```
Replacing with
```
Serial.print(F("Start  ITimer2 OK, millis() = "));
Serial.println(millis());
```
This reduces the Flash and RAM usage by 1.5 kBytes and 44 Bytes.

khoih-prog, 2020-12-17:
Thanks for your nice words and using the library.

The examples are written to cover many different scenarios and boards, with **many unnecessary displaying messages, I know, totally wasting the scarce resource of the AVR family**. The reason I did that is to **provide users as much information as possible, helping in debugging, understand what's going on, before deciding to merge into their projects.**

They are designed just to demonstrate, as much as possible, the functionality of the library. There is a **trade-off between versatility / portability and efficiency / simplicity.**

I never expect the example will be used and merged directly into the application. **Please treat the examples just examples.** There are also the possibility of some **possible problems or portability of macro F()** that I made decision then.

Those messages, helpful in debugging process, are not necessary at all, and to be removed in the actual use cases and to be determined by the users.

Anyway, your comment provides a **good guidance for novice people without much experience**, and thanks for that. I'll consider to put some notes about using F() macro or remove those unnecessary messages if you have problem with AVR memory.

If you have more constructive ideas like this, don't hesitate to post. Correct or not will benefit for everybody.

closed 2020-12-17 by khoih-prog

Django0, 2020-12-18:
Thank you for your prompt and detailed reply. It is good that you are adding a note about F(). I made a quick test and here are some numbers with the `Argument_Simple.ino`

Test1: Default: 
```
Sketch uses 7670 bytes (24%) of program storage space. Maximum is 30720 bytes.
Global variables use 562 bytes (27%) of dynamic memory, leaving 1486 bytes for local variables. Maximum is 2048 bytes.
```

Test2: Avoided all occurances of the use of ` + Sting` and without the use of the `F()` macro. For example:
```
Serial.println("Starting  ITimerx OK, millis() = " + String(millis()));
```
to
```
Serial.print("Starting  ITimerx OK, millis() = ");
Serial.println(millis());
```

```
Sketch uses 5576 bytes (18%) of program storage space. Maximum is 30720 bytes.
Global variables use 552 bytes (26%) of dynamic memory, leaving 1496 bytes for local variables. Maximum is 2048 bytes.
```

Perhaps, it would be good to also include a note about the use of `+ String`. As you can see from the above test the flash and ram usage goes down by 2k bytes and 10 bytes, respectively, without the use of `F()` macro. 

- I find your print statements very useful for example: `Serial.println("Can't set ITimer2. Select another freq. or timer");` For example, when I have exceeded the limit for Timer2.

- Also, I wondered how does an end user (like me) compute the size of your library using the Arduino IDE. I guess, I will just look at the numbers popped out after compilation. Perhaps, it would be great if you can mention something about the size of your library when enabling different timers. If you think this would be something useful for others when they choose a library.

I hope you find this useful and thanks again for your library =)

Have a nice day.
Cheers