khoih-prog/TimerInterrupt #11: No function documentation? How to use attachInterruptInterval() parameters.

Riccarr, 2021-01-11:
Hi ... there is no function documentation, only examples.  You make comments such as setting interval to KHz, such as ..
`// For 8-bit timer 2 (prescaler up to 1024, set frequency from 61.5Hz to some KHz`

... however the duration parameter is an integer type ... no decimal.  All your examples are just milliseconds. How do you set for say, 5Khz ?

khoih-prog, 2021-01-11:
Thanks for your interest in the library and also sorry for the incomplete documentation.

There are so many and more interesting works to be done, and writing docs is the **most boring / torturing jobs for library writers like us.**

I also expect the users, like you using Interrupt, is so **sophisticated** that they can read the public source code, and know what / how to use. 

That's why the documenting job is so far still in-completed.

Anyway, I'll pay more attention to improve the documentation, step by step, from now on. But please don't expect too much and too fast. I'll appreciate if you can help with the docs writing.

For you now, you can use the function **setFrequency() dealing with frequency in float**

```
// frequency (in hertz) and duration (in milliseconds). Duration = 0 or not specified => run indefinitely
bool setFrequency(float frequency, timer_callback_p callback, /* void* */ uint32_t params, unsigned long duration = 0);

// frequency (in hertz) and duration (in milliseconds). Duration = 0 or not specified => run indefinitely
bool setFrequency(float frequency, timer_callback callback, unsigned long duration = 0);
```

instead of using function **setInterval() dealing with millisecs**

```
// interval (in ms) and duration (in milliseconds). Duration = 0 or not specified => run indefinitely
bool setInterval(unsigned long interval, timer_callback callback, unsigned long duration = 0);
```

For example, if you'd like **5555.555 Hz**, use as follows:

```
ITimer2.init();

if (ITimer2.attachInterruptInterval(5555.555, TimerHandler2))
{
  Serial.print(F("Starting  ITimer2 OK, millis() = ")); Serial.println(millis());
}
else
  Serial.println(F("Can't set ITimer2. Select another freq. or timer"));
```

closed 2021-01-11 by khoih-prog

khoih-prog, 2021-01-11:
For example, if you'd like **5555.555 Hz**, use as follows:

```
ITimer2.init();

if (ITimer2.attachInterrupt(5555.555, TimerHandler2))
{
  Serial.print(F("Starting  ITimer2 OK, millis() = ")); Serial.println(millis());
}
else
  Serial.println(F("Can't set ITimer2. Select another freq. or timer"));
```

instead of

```
ITimer2.init();

if (ITimer2.attachInterruptInterval(TIMER2_INTERVAL_MS, TimerHandler2))
{
  Serial.print(F("Starting  ITimer2 OK, millis() = ")); Serial.println(millis());
}
else
  Serial.println(F("Can't set ITimer2. Select another freq. or timer"));
```

khoih-prog, 2021-01-11:
Just added [**Usage**](https://github.com/khoih-prog/TimerInterrupt#usage) to show how to use **interval / frequency.**