khoih-prog/TimerInterrupt #13: Error compiling for arduino leonardo

eslavko, 2021-02-25:
Hello...
Just entering include line in sketch #include <TimerInterrupt.h> the compiling is not possible for leonardo board

compile give:
`/home/slavko/Arduino/libraries/TimerInterrupt-master/src/TimerInterrupt-Impl.h: In member function 'void TimerInterrupt::resumeTimer()':
/home/slavko/Arduino/libraries/TimerInterrupt-master/src/TimerInterrupt-Impl.h:599:3: error: 'else' without a previous 'if'
   else if (_timer == 1)
   ^~~~
exit status 1
Error compiling for board Arduino Leonardo.`

khoih-prog, 2021-02-26:
Sorry this library now does not support **Arduino Leonardo** board.
I'll try to release a new version to add support to  **Arduino Leonardo** board within several days.

khoih-prog, 2021-02-26:
The new release has been published to provide support to **AVR ATMEGA_16U4, ATMEGA_32U4** such as **Leonardo, YUN, ESPLORA, etc.**

Please test and report if it's OK.

Your contribution is noted in [**Contributions and Thanks**](https://github.com/khoih-prog/TimerInterrupt#contributions-and-thanks)

---

### Releases v1.3.0

1. Add support to **AVR ATMEGA_16U4, ATMEGA_32U4** such as **Leonardo, YUN, ESPLORA, etc.**
2. Update examples

closed 2021-02-26 by khoih-prog