khoih-prog/TimerInterrupt #15: Compile error on UNO WiFi rev 2

trixheim, 2021-03-15:
Hi
The board check fails on UNO WiFi rev 2, so I'm unable to compile when using this board.

Best regards

khoih-prog, 2021-03-15:
Thanks for your interest in the library.

The UNO WiFi Rev 2, using ATmega4809 MPU, is different from AVR boards and is not supported by this [**TimerInterrupt**](https://github.com/khoih-prog/TimerInterrupt) or even the versatile [**TimerInterrupt_Generic** Library](https://github.com/khoih-prog/TimerInterrupt_Generic).

If there are many more requests to support this board, I'll spend some time to write the TimerInterrupt library for it.

Best Regards,

closed 2021-03-15 by khoih-prog

khoih-prog, 2021-04-02:
Hi @trixheim 

I'm just finished the initial (not sure it's working yet) version of **megaAVR_TimerInterrupt** for UNO WiFi Rev 2, but I don't have the UNO WiFi Rev 2 to test.

Are you willing to test that new library? If yes, I'll place it in private repository and invite you, so that you can download and test.

Regards,