khoih-prog/TimerInterrupt #16: Newer versions missing from platformIO

bzuidgeest, 2021-04-01:
Thank you for publishing your work for us to use. 

I tried using it thru the library manager in platformIO, but the only version it seems to know about is 1.0.2. I use an 32u4 so I need the latest version. I can copy it from the git repo, but many people might not realize there are newer versions when only looking at PlatformIO. Could you update the plaformIO registry with the newer versions?

khoih-prog, 2021-04-01:
Thanks for using the library.

For some unknown reason, PIO hasn't updated the [**khoih-prog's TimerInterrupt**](https://platformio.org/lib/show/6857/TimerInterrupt) to the latest version, so I had to create a new [**khoih.prog's TimerInterrupt**](https://platformio.org/lib/show/11520/TimerInterrupt). 

Please use the new one by changing the Author from **khoih-prog** to **khoih.prog** or to use the following command to install

```
$ pio lib install "khoih.prog/TimerInterrupt"
```

closed 2021-04-01 by khoih-prog

bzuidgeest, 2021-04-02:
I found the new version. Thanks for the fast response!