khoih-prog/TimerInterrupt #17: Adafruit feather 32u4

bzuidgeest, 2021-04-01:
I want to try your library with an AdaFruit feather 32u4. But it says its incompatible when compiling. I found this in the header
#elif ( defined(ARDUINO_AVR_LEONARDO) || defined(ARDUINO_AVR_LEONARDO_ETH) || defined(ARDUINO_AVR_YUN) || defined(ARDUINO_AVR_MICRO) || \
        defined(ARDUINO_AVR_ESPLORA)  || defined(ARDUINO_AVR_LILYPAD_USB)  || defined(ARDUINO_AVR_ROBOT_CONTROL) || defined(ARDUINO_AVR_ROBOT_MOTOR) || \
        defined(ARDUINO_AVR_GEMMA)    || defined(ARDUINO_AVR_CIRCUITPLAY)  || defined(ARDUINO_AVR_YUNMINI) || defined(ARDUINO_AVR_INDUSTRIAL101) || \
        defined(ARDUINO_AVR_LININO_ONE) || defined(Adafruit_Feather_32u4) )
  #if defined(TIMER_INTERRUPT_USING_ATMEGA_32U4)
    #undef TIMER_INTERRUPT_USING_ATMEGA_32U4
  #endif
  #define TIMER_INTERRUPT_USING_ATMEGA_32U4      true
  #warning Using ATMega32U4, such as Leonardo or Leonardo ETH. Only Timer1 is available
#else

I tried adding Adafruit_Feather_32u4 but it didn't work. Can you tell me where these defines like ARDUINO_AVR_ESPLORA can be found so I can find the proper value for my board and add it?

khoih-prog, 2021-04-01:
The **Adafruit_Feather_32u4** is not currently supported. I'll have a look to see if I can add support to that board.

khoih-prog, 2021-04-02:
Hi @bzuidgeest 

The new [**TimerInterrupt v1.4.0**](https://github.com/khoih-prog/TimerInterrupt/releases/tag/v1.4.0) has been published to add the support to Adafruit ATMEGA_32U4 and ATMEGA_328(P)

Thanks for your contribution which has been noted in [**Contributions and Thanks**](https://github.com/khoih-prog/TimerInterrupt#contributions-and-thanks)

Please test and inform if there is any issue.

---

### Releases v1.4.0

1. Add support to **Adafruit AVR ATMEGA_32U4** such as **AVR_FLORA8, AVR_FEATHER32U4, etc.**
2. Add support to **Adafruit AVR ATMEGA_328(P)** such as **AVR_FEATHER328P, AVR_METRO, etc.**
3. Update examples

closed 2021-04-02 by khoih-prog

bzuidgeest, 2021-04-02:
I have tested and it compiles fine