khoih-prog/TimerInterrupt #18: Timer3/4 in Arduino Micro board

joonghochoe, 2021-05-07:
### Is your feature request related to a problem? Please describe.

In Atmega32u4 datasheet, 16 bit Timer3 and 10 bit Timer4 can be used. But in your TimerInterrupt code, Timer3 and 4 is disabled for ATMEGA_32U4. Is there a reason to disable it?

### Describe the solution you'd like

I am working on a project using Timer3 from Arduino Micro. Just now, I am using Timer 3 by modifying the code, but is there any reason not to use it?

khoih-prog, 2021-05-07:
I'd appreciate it if you could help create a PR for this to save me some time.
I'll help on the way if you need it.

khoih-prog, 2021-05-09:
Hi @joonghochoe 

Thanks for your enhancement request, the new [**TimerInterrupt** release v1.5.0](https://github.com/khoih-prog/TimerInterrupt/releases/tag/v1.5.0) has just been published, with your Contribution noted in [**Contributions and Thanks**](https://github.com/khoih-prog/TimerInterrupt#contributions-and-thanks).

Please **help** do some rigorous tests on **Timer3 and Timer4 for ATmega32u4 and ATmega16u4** as I currently have no such boards to test yet

Looking forward to receiving more Contributions from you.

Thanks and Regards,

---

### Releases v1.5.0

1. Add **Timer3 and Timer4 support to ATmega32U4 and ATmega16U4.** 
2. Add Timer auto-selection to examples.

closed 2021-05-09 by khoih-prog

joonghochoe, 2021-05-11:
When I used version 1.5.0, Timer 3 worked fine. Thanks.

khoih-prog, 2021-05-11:
Good news.If you have time, could you please help testing Timer4. Thanks.