khoih-prog/TimerInterrupt #19: Leonardo gives old error in #13 when using in Atmel Studio

utsavZebra, 2021-05-11:
### Describe the bug

Compiler complains about `'else' without a previous if` whereever `#if ( TIMER_INTERRUPT_USING_ATMEGA_32U4 )`  is
in TimerInterrupt-Impl.h

The examples provided in Arduino IDE work though. 

### Steps to Reproduce

An atmel studio project with .h files moved from src_h

### Expected behavior

A clear and concise description of what you expected to happen.

### Screenshots

[image]
[image]
[image]

### Setup
```
Microchip Studio: 7.0.2542
Arduino Leonardo: ATMega32U4

```

khoih-prog, 2021-05-11:
Hi,

Thanks for using the library.

I just retested the examples and they're compiled OK, **without any error for Leonardo 32u4.**

[image]

I'm afraid you did **something wrong in your code or installation.**

I suggest you

1. Try with the simple example first to see if it's compiled OK on Arduino IDE.
2. Then move to the new Microchip Studio to see if it's still OK. Even if this is not OK, I still won't work on the issue as this is the responsibility of other people.

You also had to post the [**MRE**](https://stackoverflow.com/help/minimal-reproducible-example) at least and to prove this is **the issue caused by the library**, so that I'll re-open the issue.

Good Luck,

utsavZebra, 2021-05-12:
Update: Not a problem with the library. Atmel studio has the option to make a project from an Arduino Sketch. That did the trick as opposed to adding Arduino library in an existing project. Sorry about the confusion and MRE.

khoih-prog, 2021-05-12:
Good news. Thanks for the info.